TBXin/NFirmwareEditor #1: Images getting messed up by resizing

Husky110, 2016-03-28:
Hello,

Thanks for the awesome tool.
I was trying to recreate the fonts from Super Mario World into my eVic VTC.
To do so, I had to resize some of the Images from 5x8 to 6x8.
Something seems to go south when reencrypting the final file, since I encounter broken pixels all the time (which have been moved or switched) and when I open a saved (as encrypted) file, some of the images are broken.
I'll attach an encrypted file, so you could see what I mean (had to zip it, cause of github).
Within the file: 
 - The 1 pixel on 0x01 on the top right wasn't created by me...
 - On 0x54 you should see what I mean...
Could you please look into that?

TBXin, 2016-03-28:
Hello,

Thank you for feedback.
Usually, images data follow one by one, and each image takes specified bytes count, which is calculated by its width and height.
If you want to resize an image in a big way, then you need to realize that it is will corrupt data of the following image(s).
Simple and automated way to resize in a big way does not exist. Currently only one way to do so - create patch which will move image data to another place in the firmware and make corrections of the image table.

Husky110, 2016-03-28:
Thanks for your fast response.
I just figured out that this (like you said) when I try to enlarge an image, but NOT when I make it smaller (like from 10x10 to 8x8).
And I figured out that the firmware itself comes with a huge overhead of unused images.
So what I will do is to set up some sort of "map" which images are natively used by the eVic (since I have no other device) and just use the overhead to achieve my goals.
I will link that map as soon as it's ready. :)
Still thumbs up for the tool itself. :+1:

TBXin, 2016-03-28:
You will do an amazing job, if you're done with the map.
Regarding unused images - them not so much as you think. I am sure about E1 E2 and E3 images in both blocks.
Also, two blocks with images is not just for fun :)
Various hardware versions uses different LCD screens, and one type uses block1 images, when second one uses block2 images.

Husky110, 2016-03-28:
I'm having here an eVic with Hardware Version 1.04 and therefore I'm creating the map for, maybe someone will map the other devices too, so we all could have some fun.
If you wanna see my progress so far in modding the eVic, check out www.husky110.de/uploads/eVic_VTC_Mini_V3.01-mario-TC-FINAL.bin

Husky110, 2016-03-28:
I'm done mapping the images for the eVic Hardware 1.04 and Firmware 3.01.
It's available at: www.husky110.de/uploads/Imagemap.ods
I'll update a Stringmapping within the next few days. :)

TBXin, 2016-03-28:
Great Job :)

Husky110, 2016-03-28:
Stringmapping is now in there too. Might not be complete (not enough time today ;) )

Husky110, 2016-03-29:
When I'm done with my map, I am going to see if I can map other devices too.
As soon as this is ready, I will put it online.

closed 2016-03-31 by TBXin