TBXin/NFirmwareEditor #5: Compatibility

Insane88, 2016-04-15:
Hi,

Does this software works on Cuboid with H/W version :1.2 ? Because i cannot seem to make it work , no matter what i try.

Thank you!

TBXin, 2016-04-15:
Hi,

This application can edit the firmware and load them into the device.
What exactly do you want to do?

Insane88, 2016-04-15:
edit/update/modify the font. or apply patches. I should mention i tried on my friends Cuboid H/W version 1.01 and worked perfectly.

TBXin, 2016-04-15:
I do not know about which patches you are talking about, but there is almost no patches for cuboid.
Also, you need to know, that there are two image blocks in the firmware. 
And most likely you are modifying, only the first.

Insane88, 2016-04-15:
It may be my fault for downloading a .bin file from the internet and expect it to work on my mod, but i was a little baffled when it worked on my friends mod ( HW 1.01) https://www.youtube.com/watch?v=Uqd4PwbE_RI 
Thank you for your input so quickly , i will try as suggested and edit both fields first.

Insane88, 2016-04-16:
I have managed to successfully change the font by copying the first images block to the second on the firmware listed on this site http://pictape.com/devices/custom-firmware-for-the-evic-vtc-mini-cuboid-rx200.php

Thank you !

closed 2016-04-16 by Insane88