TBXin/NFirmwareEditor #7: rx200 Bitmaps

applehat, 2016-04-18:
The editor doesn't appear to allow the edit of the 0xA block of bitmaps.

There is a patch available for the RX200 3.1 firmware that says "show the 0xA2 splash screen" but I can't actually modify the 0xA2 block.

TBXin, 2016-04-18:
Hello,
Can you explain in more details?
I tested this patch and successfully changed the image for it.

applehat, 2016-04-18:
I apologize. I just realized this is an issue on my end.

I have been running the tool inside Wine on OSX - and it works great, but the block selector doesn't seem to scale correctly in Wine (not an issue with your code).

[image]

I realize now if I use my keyboard I can keep going down to more blocks (off screen).

[image]

You can disregard this issue. Sorry again.

closed 2016-04-18 by applehat

TBXin, 2016-04-18:
No problem :)
I recommend you use Parallels Desktop instead of Wine. Coherence mode is awesome!

applehat, 2016-04-18:
At home I have a Virtual Box instance with windows installed, but at work I dont... so I decided to see how it worked with wine =).

maelstrom2001, 2016-04-18:
`winetricks dotnet40` and everything will work like a charm )