TBXin/NFirmwareEditor #8: RX200 broken after Flash with internal Firmware Updater

hitme666, 2016-04-18:
My RX200 is broken after flashing it to 3.10 with internal Firmware Updater.
My my dosn't reconioze it as USB device anymore, Display stays black also, no matter what i try.

pc only showing "USB\DEVICE_DESCRIPTOR_FAILURE" in device managment.

TBXin, 2016-04-18:
Hello, 
Probably are you're somehow damaged the firmware file, for example - trying to insert a larger image than it was.

You need to take off batteries from your RX, press PLUS button and hold, then connect to usb and release plus button. Device will identify with FW version: 0.00.
Now you can upload correct firmware file.

hitme666, 2016-04-18:
Hi, that helped!

Is there a way thats the internal Firmware Updater checks the firmware before flashing it?

TBXin, 2016-04-18:
Unfortunately no.  NFE can't validate firmware code blocks.
NFE validates only that firmware is you trying to upload is satisfy device.

hitme666, 2016-04-18:
wenn, anyway thank you for all your affords! ;)

closed 2016-04-18 by hitme666