TBXin/NFirmwareEditor #9: Change the default 15s time on the splash image?

KevinACrider, 2016-04-19:
Looking at the patch for the splash it says the default time 15s and on line 17 it appears you can set the display time. First off, does this mean the time at which it comes on? And if it does, what values need changed what do they correspond to? I tried converting the various hex values to text and to binary but none of them seem to indicate anything close to 15 seconds.

Additionally, is there a way to make this splash screen show when the device is woke up from a blank screen, for 1 or 2 seconds, prior to showing the information? Or even just when it's turned on after 5 clicks?

Thanks so much for your hard work! This is absolutely incredible!

maelstrom2001, 2016-04-19:
Thank you for the appreciation of our work!
All "Splash" patches are based on countdown time to turn off the screen. This means that the mod is counting seconds from last key press to 0, and then screen will turned off. Default time for countdown is 30 s for all mods, and "Dim" patches changes this value to 10/15 s.
"Splash" patch contains time value when the splash should be displayed, it is the same - the duration before display will be dimmed. So, by default splash screen will be shown for a half of total timeout. If you want to increase this time, for example, to 20 seconds, change corresponding value in patch file to `14` (hexadecimal representation of 20), and splash will be shown after 10 seconds of inactivity and for 20 seconds in total. Also, note that this value cannot exceed the primary timeout value (30 s without patches, 10/15 with "Dim" patches installed).
For the second part, displaying of splash during on/off perhaps will be implemented by another patch.

KevinACrider, 2016-04-19:
Man, you are so awesome! Thank you so much for that in depth and perfect explanation!

I will keep my eyes out for another patch and continue to be envious at how these patches are created in the first place.

closed 2016-04-20 by TBXin