TBXin/NFirmwareEditor #11: Editor need a fix

StormtrooperBF, 2016-04-20:
With the new firmware it is possible to place pictures onto the Cuboid and the VTC-Mini interface. However these logos are hidden somewhere in the firmware, which cannot be retrieved by an editor to edit them. Can you please fix this?

TBXin, 2016-04-20:
Hello,

Logo stored not in the firmware.
Logos are located in the memory area, which is outside the firmware area.

So, you never can't edit they while edition firmware file.

StormtrooperBF, 2016-04-20:
Is it possible that we will receive an export function so we can edit the pictures as usual and save them as bitmap? That is the only thing i see which could make it possible.

TBXin, 2016-04-20:
You can use Joyetech Updater Software to upload a custom image.
There is no way to download previously uploaded image.

bluenazgul, 2016-04-20:
I belive what StormtrooperBF mean is an "Export to BMP" inside NFirmwareEditor so we are able to extract the Logo we want from older Firmware we created an will be able to use this Image with the Joyetech Updater

StormtrooperBF, 2016-04-20:
This is exactly what I mean.
Thx bluenazgul

TBXin, 2016-04-20:
Okay, this is possible. I will add this idea to the NFE todo list.

jd0yle, 2016-04-24:
Seconded... would be a really handy feature

TBXin, 2016-04-24:
Look at 3.71 release.

closed 2016-04-24 by TBXin