TBXin/NFirmwareEditor #14: Old vtc mini 3.01 patches

delchrys, 2016-04-24:
Hi,

how can i use my old 3.01 patches with the new 3.02 firmware.
I would like to use the display brightness and some other patches, but they are not compatible anymore.
How can i fix this???

TBXin, 2016-04-24:
Hello,

You can't. Patches are created exclusively for a specific firmware.
If you know assembler, you can port your own. But it seems to me that you do not know it :)
So, i'd recommend to you rollback your device to 3.01 firmware.

delchrys, 2016-04-24:
i'll have to wait then until someone ports them to 3.02.
or i have to figure out the new adresses so i can rewrite the patches

TBXin, 2016-04-24:
it is your choice, so i can't decide for you.
Also, patch-makers that i knew tell me, that they will not porting patches from 3.01 to 3.02.

delchrys, 2016-04-24:
why don't they port to 3.02??

TBXin, 2016-04-24:
No sense.
3.02 and 3.01 are same for end-user, except custom logos.
But for 3.01 already has been patch, that doing same things.

delchrys, 2016-04-24:
i liked the 3.02 function of selectable logo on/off. combining the temp modes. And the logo instead of POWER and still showing watt when button pressed. 3.01 patches don't have those options.

TBXin, 2016-04-24:
I can't help you. I don't create patches.

delchrys, 2016-04-24:
ok thanks for the info anyway

closed 2016-04-24 by TBXin