TBXin/NFirmwareEditor #16: Evic Vtc Patches

MoDuLah, 2016-04-24:
After trying some of them. I've seen that many are not compatible even with standalone.
Would be nice to see them updated for 3.02 firmware.

TBXin, 2016-04-24:
https://github.com/TBXin/NFirmwareEditor/issues/14

MoDuLah, 2016-04-24:
no no no.. i mean skip modes for example is not compatible on a 3.01 firmware.

TBXin, 2016-04-24:
I gave you a link for this reason:
"patch-makers that i knew tell me, that they will not porting patches from 3.01 to 3.02."
"I don't create patches."

All patches for 3.01 are compatible with 'stock' firmware. If it is not true for you, that means, that you have not the original firmware file.
[image]

MoDuLah, 2016-04-24:
ok thank you for your help.

closed 2016-04-24 by MoDuLah