TBXin/NFirmwareEditor #17: Software crashes when previewing respack for Cuboid

TheHamster95, 2016-04-24:
When trying to preview respacks for the Cuboid, firmware 3.01, I get an error-message stating "Sequence contains no matching element." Details of the crash are 

************** Exception Text **************
System.InvalidOperationException: Sequence contains no matching element
   at System.Linq.Enumerable.First[TSource](IEnumerable`1 source, Func`2 predicate)
   at NFirmwareEditor.Windows.PreviewResourcePackWindow.GetImageByIndex(Int32 index)
   at NFirmwareEditor.Windows.PreviewResourcePackWindow..ctor(Firmware firmware, IList`1 originalImageIndices, IList`1 importedImages, Boolean resourceImport, Nullable`1 defaultImportBlock)
   at NFirmwareEditor.Windows.Tabs.ResourcePacksTabPage.PreviewResourcePack(ResourcePack resourcePack)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Happens in latest version, 3.71 on 2 machine both running Windows 10 Pro.

TBXin, 2016-04-25:
Thanks for your feedback.
I wll add protection in the code for this case.

First of all: Cuboid 3.01 is not supported. (or you have custom definition file or editor determines 3.01 as 3.10 and it is wrong)
Secondly: error occurred because resource pack has some glyphs that does not exists in the firmware. Most likely it happened because it is not designed for 3.01

TBXin, 2016-04-28:
Fixed in the 3.8

closed 2016-04-28 by TBXin