TBXin/NFirmwareEditor #21: [Suggestion] Preview Tab

MoDuLah, 2016-04-26:
Would be nice to have an extra tab after/before resource packs with dummy screen info but with all the possible screens visible. This could also help the screenshot section but also would see instant changes.

ganthamkm, 2016-04-29:
For some reason with the NFirewareEditor v3.8 this is an issue with the splash 2. Im not sure is the issue it with v3.71 or not cuz i havent tried to use it but the issue is on my rx200 it won't charge through usb. If you have just the regular wismec firmware with you pull it up while it's off the charge bar flashes to show it charging but with the editor on splash 2 it doesn't do that and it doesn't charge and I know for sure it doesn't charge because I left it on over night to see if it just didn't flash anymore with splash 2 added but that wasn't the case it just didn't charge at all. Please fix this or let me know if I did something wrong or w.e the case may be. Thank you.

TBXin, 2016-04-29:
Hello, ganthamkm.

First of all thanks for your feedback.
Firstly: NFE - Firmawre editor software. Patches is not a part of the NFE;
Secondly: Yes, this is a bug in the splash 2 patch for RX200;
Thirdly: i do not develop any patches or support them;

ganthamkm, 2016-04-29:
I'm new to making them myself. I just noticed this issue . But so your saying there is no way for you to actually fix the bug ?

MoDuLah, 2016-05-02:
ganthamkm ur off topic.. make ur own..

TBXin, 2016-05-02:
MoDuLah, preview is a cool feature, but very hard to implement. 
There are a lot of patches that changing primary and other screens. In this case editor can't give you a fair preview.
To create fair preview we need to implement fully working virtual mod, like virtual machine for OS. This is a very complex task.
Therefore, this issue likely will not be implemented.

ganthamkm, 2016-05-03:
Um MoDuLah no need to get Smart  like i said I'm new to. I wasnt talking to you in the 1st damn place. Thanks

ganthamkm, 2016-05-03:
Thanks for breaking the down TBXin.

closed 2016-09-05 by TBXin