TBXin/NFirmwareEditor #22: Linux/Wine  Support

LucyLestat, 2016-05-04:
First off, Great project! Love modding my vape :) Would love to see some Linux support, or maybe some way to make it Wine friendly. Idk if its just me, but absolutely no luck with Mint 17.2. Program (menus, screens, load dialogs) all seem to work, but doesnt recognize files, at all. tried on Wine's C: drive and in ./home folder, nothing. tried WinXP, Win10 and ReactOS in Virtualbox, same results. I'm very anti M$ and would love to use this program. many ideas and would love to contribute. 
Thanks and keep up the good work xD

TBXin, 2016-05-04:
Hello,

Thanks for your feedback.

In general, software designed for os windows, but it can be used at wine or virtual machine with installed os windows.
Perhaps you are using old version of wine or something else. 

I saw many screenshots in the web with NFirmwareEditor that's running on wine.

For better experience i'd like to recommend you install virtual machine, such as: VMWare, Virtual Box or something else, then install windows 7/8/10, install .NET Framework 4.0/4.5/4.6.

Musik234, 2016-05-04:
What do you mean with
> but doesnt recognize files, at all.
I run NFE in a wine-window on Linux Mint 17.1 and it works fine exept some minor optical issues. I also cannot upload the firmware to my eVic but thats because of the poor usb support in wine. I use [python-evic](https://github.com/Ban3/python-evic) to update my box.

LucyLestat, 2016-05-05:
first off, holy shit that was a fast response! :) 
TBXin:  not trying to be a smart ass, but I definetely noted i tried multiple M$ OS(s) in Virtualbox. im pretty sure my mint 17.2 and Wine are both up to date, and so is my VB install. Maybe I just didnt look in the right spot(s), but I'll beleive SOMEBODY must have this running on *nix/Wine. As for installing a M$ os, avoiding doing that was my entire reason for even making this thread. Unless somebody can point me in the direction of a free AND *legal*  version of Windows, I'm kinda screwed. Again, apologies for being a smart ass. 

Musik234: The issue I seem to be having is that the program doesnt seem to see or even acknowledge there any usable files anywhere on my hard drive. for example, my pictures are in (the usual) " ./home/Pictures/ " directory. however, NFE doesnt list ANYTHING at all being in that folder. even when i stick both the original firmware for my rx200 and both .bmp and .jpg images inside the NFE folder itself, still nothing. If I cant select the firmware I wish to mod, then I cant actually load the firmware to mod it. same thing occurs on win10 and winXP when i run virtualbox. even if I put the file inside the virtual machines virtual hard drive I still seem to not get any options. 

hope I'v been verbose enough, if screenshots or such are needed, let me know and I'd love to help. And if I'm making a simple mistake, please correct me lol

TBXin, 2016-05-05:
LucyLestat, obviously, the problem is in your environment.
Last night, Just for fun, i have installed ubuntu, wine, winetricks, .net 4.5 and mono.
After all this simple steps for a unix/linux user i can run NFirmwareEditor and it works almost perfectly.

Yes, there are few issues, but this is wine problem, not mine.
Firmware Updater - doesn't work;
UI lags and some controls does not worked as designed;

But, i can open, edit and save firmware file without any questions and advanced manipulations with wine or something else.

Unfortunately, I can not longer help you, since i am not a user of Linux and do not know any advanced wine settings.

LucyLestat, 2016-05-05:
*Update*  upon further inspection, I had come to realize that for some reason, my system was still running Wine 1.6, I upgraded to Wine 1.8 and pretty much everything seemed to have worked. Noticed the lag you speak of, and doesnt display window contents properly (some objects in list dont appear on screen, splash screen grid stretches way too fat, etc). So far everything else in your program appears to be functional. (currently working on a personal theme) 

As for the virtualbox, havent tested it since I updated Wine. I'll have to try again when it comes time to flash my firmware over to my mod. 

thanks for testing that for me and hope you enjoyed your Ubuntu experience :) Look forward to some awesome shit to come from this.

closed 2016-05-05 by TBXin