TBXin/NFirmwareEditor #26: [Suggestion] Blocks 1, 2, 3?

yvanovsk, 2016-05-05:
Hi!

I might be mistaken, but since the last firmware updates with the FlappyBird game it might be possible that there is a new image block added to the firmware. If you did/do/will investigate this possibility, then there might be a need to work with more than 2 Blocks.

Thanks in advance,
Regards.

TBXin, 2016-05-08:
Hello,

Possible, once it will be need when Joeytech/Wismec/Eleaf start making devices with display types, not 1306 and 1327.

Block 1 represents images for 1306 display.
Block 2 represents images for 1327 display.

Each block has own image format.

yvanovsk, 2016-05-08:
Oops, my bad - I forgot about that.
I'll try rephrase: since there is a game included in the firmware, might it be that game resources are not read by NFE because they are not included in a common resource table? Might there be a third/fourth resource table special for the game?

TBXin, 2016-05-08:
There are two image tables: for block1 and block2. No in one of them game related images are not presents.

So, or firmware draw each sprite per pixel (hardcode) or sprite images are not located in the tables.
Both variants are not good for us.

I am inclined to think that the firmware has no game related resources (sprites).

yvanovsk, 2016-05-08:
Ок. Понял. Спасибо.
Ok. Got your idea. Closing this issue till someone will dig all the game code.

closed 2016-05-08 by yvanovsk