TBXin/NFirmwareEditor #29: Can't find Cuboid 3.13 firmware

Elizzle, 2016-05-08:
In the update log on this version (NFE v3.81) I see it says added support for Cuboid v3.13, and when I open the Firmware Updater tool it does read that the mod is running v3.13; however I can't find a 3.13 respack or patches folder anywhere within the program. Is it only compatible to read it, not modify as of latest update? Thanks, any help is appreciated.

TBXin, 2016-05-08:
NFE - Firmware editor software. Patches, resource packs and so on is not a part of the NFE.
NFE can open, edit and save Cuboid 3.13 firmware files.

[image]

Elizzle, 2016-05-08:
I noticed that, I thought you could install third party created patches through the NFE program onto v3.11+3.12? How would I go about implementing patches into the v3.13 firmware file? All I see is folders for the older firmwares. Thank you :)

TBXin, 2016-05-08:
You can install patches and resource packs to 3.13/3.02 firmwares, but there are no patches/resource packs.

So, you can only wait when they come :)

Elizzle, 2016-05-08:
Oh, that makes sense. Thank you again!!

closed 2016-05-08 by TBXin