TBXin/NFirmwareEditor #32: Save As... Behaviour

yvanovsk, 2016-05-12:
Hi,

The "Save As" behavior is not as default behavior in Operating System or Office products:
In Windows when you do "save as" the currently open file is saved with a new name and the editor starts working with this file.
I NFE, if you do Save As - a copy of currently open file would be saved with a new name, but editor will continue to edit the original file, not the new one. This behavior is equal to "Save a Copy" in Notepad++

This difference in behavior is a bit frustrating.

Would be nice to fix the behavior and, if possible, add the "Save a copy" function also.

TBXin, 2016-05-18:
Hi,

Thanks for your feedback.
Look at the release 3.9.
Now Save as worked as expected.

yvanovsk, 2016-05-18:
Nice! Thanks!

closed 2016-05-18 by yvanovsk