TBXin/NFirmwareEditor #36: Multilanguage support

LaMoLa, 2016-05-25:
Congratulation for this great software
.- Will You be so kind give us a possibility to translate it to  other languages in txt or in xml file make Your application more comfortable for others. 
- Is there any way to add special characters like :  á Á é É ó Ó ö Ö ő Ő ú Ú ü Ü ű Ű  Thank You for Your attention.
Best regards from Budapest Hungary

TBXin, 2016-05-25:
Hello,

Multilanguage UI is a very complex task. I will think about this feature.

About special characters: if you mean adding them to the firmware - you can redraw english glyphs for your own and then edit strings.

LaMoLa, 2016-05-25:
Thaks for the quick answer
I've translated the iStick TC100 to Hungarian but I can redraw english glyphs which are not in work. We have more letters in the hungarian lang. If You can add some empty it can be fine  We have 16 more letters as You see.
Thanks for Your attantion.

TBXin, 2016-05-25:
Unfortunately, there is no easy way to add additional glyphs in the firmware.
To implement this you need to write a patch, which will allocate space for each new glyph and extend image table with new offsets for each new glyph.
I can not help you with that problem, since i am is not engaged in writing patches.

mirko93s, 2016-08-24:
How can I add mor byte to a string? Some words in my language are longer

LaMoLa, 2016-08-28:
HI Mirko

There are lome letters not used in the text file. You can modify these letters.

These letters ar  P  Q  X  Not enough for our special fonts,  but not too bad

P->Ő  Q->Ö  X-> Á  U-> Ú    It was enough for  iStick TC100  and  Pico  in Hungarian - Some word longer in HUN  Low  etc. These are  still in  English  [☹]

The author does not want to work with this problem. Too much work to make similar size than  Battery and write the letters in that area. Not only 1 letter in 1 letter box.

That's all I could make with it.  Good  luck.

LaMoLa from Budapest.

ps. I cannot add the newest Pico to the XML file.  Does not work [☹]

________________________________
Feladó: mirko93s <notifications@github.com>
Elküldve: 2016. augusztus 24. 23:56
Címzett: TBXin/NFirmwareEditor
Másolatot kap: LaMoLa; Author
Tárgy: Re: [TBXin/NFirmwareEditor] Multilanguage support (#36)

How can I add mor byte to a string? Some words in my language are longer

ZelTroN-2k3, 2016-10-22:
bonjour,

je me suis penché sur le sujet posé par "LaMoLa", et je me suis permis de convertir "NFirmwareEditor
 5.1", parcellement en français.

voila quelque images...
[image]
[image]

perervin, 2016-11-12:
Bro! You may add russian (cyrillic) font АБВГДЕЖЗИКЛМНОПРСТУФХЦЧШЩЪЬЭЮЯ абвгдежзиклмнлпрстуфхцчшщъьэюя

ZelTroN-2k3, 2016-11-15:
vola mon travail actuellement
[https://github.com/ZelTroN-2k3/NFirmwareEditor](url)

closed 2016-12-17 by TBXin