TBXin/NFirmwareEditor #37: Newest versions not working on Windows/Can't open firmware 3.13 Cuboid

erickristof, 2016-06-02:
On windows 7, I can run version 3.81 just fine, but neither 3.9 or 4. When I try to run the exe I immediately receive a "nfirmware editor has stopped working" error message.

In version 3.81 I cannot open the latest firmware for the Cuboid, v3.13. I get a message saying "Unable to load firmware. No one definition is not appropriate for the selected firmware file". I'm not sure if this is to do with the version of NFirmwareEditor I'm running, as I can't run the newest version.

TBXin, 2016-06-02:
Can you attach log.txt file to this issue?

erickristof, 2016-06-02:
Not sure if I did this correctly, and there appeared to be 2 related event logs under Event Viewer. If this doesn't help let me know what I can do to help.

[log1.txt](https://github.com/TBXin/NFirmwareEditor/files/296630/log1.txt)
[log2.txt](https://github.com/TBXin/NFirmwareEditor/files/296631/log2.txt)

TBXin, 2016-06-02:
I mean log.txt file, that is located near NFirmwareEditor.exe file.
Also, you can try to remove settings.xml and then rerun applicaiton.

erickristof, 2016-06-02:
Thanks so much, that worked! I can open the program now.
Did you mean the Nlog.dll file ? I don't see any .txt files. Sorry I'm not experienced with this stuff so I'm probably not able to help. But I can use the program now, so that's awesome!

TBXin, 2016-06-02:
Nope, i mean log.txt :) 
So, now it is doesn't matter.
Thanks for log files that you are attached earlier.

Have fun with editing firmware files.

closed 2016-06-02 by TBXin