TBXin/NFirmwareEditor #39: Changing string length

01zabil01, 2016-06-09:
Hello, I really respect you for this software editor! One thing i've had problem with is changing string length. I've figured out how to do this, but it was time consuming tho. I've found some unused chars and just changed them to full text bitmap made by me. For example make text in any photo editor HELLOWRLD, change it to bitmap, find unused char X, change X char to HELLOWRLD bitmap and then go to string, find POWER and change
P
O
W
E
R

to 

X char (HELLOWRLD)
NULL
NULL
NULL
NULL

It works but it takes time. Are you planning to implement string length editor or smth like that?

TBXin, 2016-06-09:
Hello,

I don't recommend to you increase glyph sizes since all glyphs are arranged one by one. Increasing size of the first you automatically ruining second glyph. Also, original size will be lost after you save firmware.

Situation with strings is exactly the same.If you want to reduce string size, you need just select NULL in the desired place.

P
O <-- Put NULL HERE
W
E
R

and you will get

P
NULL
NULL
NULL
NULL

01zabil01, 2016-06-09:
Nah, you didn't understand my problem :D. My problem is I want to change 5 chars string into 6 chars string, so it gets complicated then. Is there going to be some string length editor?

01zabil01, 2016-06-09:
Also, is there a way to add chars? I want to make custom icons, but i can't find any more unused chars. And 1 more. Will Flappy Bird textures be available to edit?

TBXin, 2016-06-09:
> My problem is I want to change 5 chars string into 6 chars 
See my answer about glyphs and strings.

> Will Flappy Bird textures be available to edit?
There are no textures. All game sprites are drawing programmatically.

closed 2016-06-12 by TBXin