TBXin/NFirmwareEditor #40: [REQUEST] Remove battery check on RX200S firmware?

scottstamp, 2016-06-10:
Hey there, I know it's still in the works pending an initial upgrade from Joyetech/Rouleaux, but I'm trying to modify a 200S to run from a power supply.

Currently the board checks the voltages on the balance leads connected to the sled and refuses to fire as it cannot detect the individual voltages for the batteries (reads "Check Battery"), but the board does power on, so I'm thinking this is strictly a firmware limitation.

It'd be really awesome if you guys could work this into an upcoming release when the files for the 200S are available, if it's not too much trouble!

maelstrom2001, 2016-06-11:
Hello!
Did you try to implement schematic like this?
[image]
Values of the resistors are just for reference. I suspect that RX will work without firmware hacks because there will be voltage on checked cells terminals.

scottstamp, 2016-06-12:
Yeah, I was planning to mod it like this using some 1K trim potentiometers to get the values dialed in, but a firmware solution would make it significantly easier. I'm helping a friend with a project which may include a small production run of passthrough devices based on the RX200/S chipset and having a firmware solution would simplify manufacturing.

scottstamp, 2016-06-13:
Looks like Wismec released a copy of the firmware today! Doesn't seem to be an update, just a copy of the stock firmware (maybe it was released to restore to stock, or update early production models?)

http://www.wismec.com/software/wismec-tc-software-for-upgrading-reuleaux-rx200s/

I'd edit it manually, but I have no idea how to find the offset addresses to decrypt the firmware. Either way, it's a start.

TBXin, 2016-06-13:
Hello,

Thank for information.
Just released NFE 4.02 with RX200S firmware support.

scottstamp, 2016-06-15:
I was able to implement iNFiNUM's patch to change the cutoff voltage on the RX200S Firmware (v4.00, offset 000006B2), but it doesn't seem to actually disable the battery check. Thanks for the update though, at least it's a starting point.

closed 2016-08-28 by TBXin