TBXin/NFirmwareEditor #41: Cuboid 3.12

MYWARUK, 2016-06-12:
Hi there, 
When I had my Cuboid few months back, Before I sold it on, I was able to edit the official 3.12 firmware and able to use Patches and Resource Packs. I got another Cuboid, downloaded the 3.12 firmware, but now no Patches or Resource Packs are there, When I try to load them i get the message:

"Selected resource pack is incompatible with the loaded firmware.
Resource Pack is designed for: Cuboid 3.12
Opened Firmware is: Cuboid 3.12 (31.05.2016)"

If someone has a link to the older 3.12 that be great

Cheers for your work, great tool!

TBXin, 2016-06-12:
Hello,

Here you are: http://rgho.st/7DqPjfpxS

MYWARUK, 2016-06-12:
Thank you so much!

I'll keep that backed up.

Thank you again 👍

closed 2016-06-12 by TBXin