TBXin/NFirmwareEditor #43: [Question] Testing hex edited firmware

SaschaK, 2016-06-21:
Hi,
I'm a software developer, too.

Is there any possibility to test a firmware, that is edited in an HEX-Editor? Or how do you find the offsets? I viewed your source, but I didn't work with such kind of hardware befor.

TBXin, 2016-06-21:
Hello,

> Is there any possibility to test a firmware, that is edited in an HEX-Editor?

Upload to device.

>how do you find the offsets? 

With my eyes :)

> I viewed your source

So, now you know how and where the data is read.

SaschaK, 2016-06-24:
Ah.... Bad answer. In you don't want to help others to understand your work you shouldn't publish your Code.
Do you decompile the original fw to find the offsets? Which tool you use to do it? 

I only want to understand it, because I want to try to create a patch by myself.

closed 2016-07-04 by TBXin