TBXin/NFirmwareEditor #49: [REQUEST] patch for rx200s

RenegadeTitan, 2016-07-20:


olokos, 2016-07-23:
+1 on this. I was going crazy with 0.4mm SS316L wire that I used to make coils. Each one of them did that sooner or later. Especially on 0.07ohm resistance. One of workarounds is setting tcr manually to 170 in M1 mode, another is blaming Wismec for messed up SS mode that cannot handle 316L properly.

RenegadeTitan, 2016-07-24:
changing tcr helps to make temp readings more accurate for sure due to the device reading ohms about .01-.02 higher than they are. raising temp effectively alters performance the exact same way as raising tcr. as long as u dont hit 600f limit. but unfortunately doesnt affect the switching to vw mode at all. Or affect the 30-40w coil kanthal check (effectively a super slow pre-heat) that causes this switch at all. thank you for your +1. I know many many others are apparently frustrated by the switching to vw mode. As far as i can tell i am the first to notice its caused by the strange 30-40 watt kanthal test range. I am really hoping this observation will help TBXin to pinpoint the location of code to change, making these devices function with heavy ss builds. This issue is really driving me crazy. TBXin youll be my personal hero if you can sort this out :)

TBXin, 2016-07-25:
Sorry guys, patches is not my prerogative.
Almost all patches was done by maelstrom2001, but at this moment him sold all joyetech/wismec devices and uses DNA200 like me.

Sometimes i am looking for firmware updates and adding support of them to the NFE, but this is all that i can do now.

RenegadeTitan, 2016-07-26:
I solved it! thank you for your response. i dont ask others to try to do anything wouldnt do so ive been working on it for a few days. man that was a lot of work to find and change 2 bites lol. this changes 30-40w range to 30-249.6w range. at .23 24g/30g clapton works amazing now in tc mode. depending on if its a % of original resistance or set ammoint of ohms could still be a prob at much lower ohms. ill see about that too. i dont think lower limit is causing any problems so didnt lower it to 1w, but raising it could be a pre-heat for first .03 ohm change or so. would be good on ss since that has small ohm range anyway. i can make a patch for that too. ill post rx200 patch like this one when i make that just to help out those mod owners pulling their hair out like i was. i couldnt have done it without NFirmwares ability to save decrypted option so thank you TBXin. ill post patches as i make them now that i figured out how to do this. here is the life changing patch i made lol (just rename extension to .patch of course)
[tc_SS_fix.txt](https://github.com/TBXin/NFirmwareEditor/files/382800/tc_SS_fix.txt)

TBXin, 2016-07-26:
@RenegadeTitan greate job!
Just added your patch to the NFE patch repos, and now every NFE user can download it when opening RX200S 4.10 firmware file.

Of course, you can send patches to me directly, or do pull requests to the repository.
I am very glad that there are people who help develop the project.

kushpancake, 2016-07-27:
Hey Reiko,

I just got my first dna device (SILO 2000) but can't run the escribe
software being all I have at the moment is a MacBook air. Unfortunately
installing VMware fusion, parallels, or bootcamp on my air will definitely
take up too much memory. Im currently running your NFirmware app using
Wine, but that doesn't work with the escribe software. Do you possibly
happen to have any sort of work around for me??? Thanks in advance!!!!!

TBXin, 2016-07-27:
Hello @kushpancake,

Unfortunately for you there are no Escribe version for OS X. 
Probably best solution for you is creating virtual machine using virtual box, installing windows 7 and run on them Escribe software.

Also you can look here for simple guide how to: OS X + Virtual Box + Escribe.

olokos, 2016-07-27:
@RenegadeTitan I would really appreciate it if you could also do this for RX200. I'm a big fan of SS and I've been going crazy with this thing switching to VW almost making me to give up SS whatsoever. In fact I'm mostly doing kanthal coil just because of the fact that it will work everytime while SS might now. My problem would be gone if you could do that. Thank you! :smile:

RenegadeTitan, 2016-07-28:
TBXin i added 149.6w preheat to last patch by changing minimum wattage of kanthal check. notes inside patch explain all the benefits. more than i thought at first. ss mode works perfect. (also tested thinner titanium coil and preheat worked great.preheat switches to tc wattage quickly as soon as ohm changes a few points. downside (like all high preheats i believe) is that this one has potential to fire at 149.6 watts for first full 2 seconds of pushing fire button if using kanthal in tc mode or if base coil resistance is set while coil is hot. in these 2 circumstances then after 2 seconds it would fail check and switch (correctly) to vw mode at 20w. i put warnings in description. wanted to let you know that it has these potential downsides so you can choose to not include it if you feel those 2 seconds at 149.6 watts if user makes those 2 mistakes is too dangerous. it dramatically improves ss mode. im using it and i made it for rx200 too. 
rx200s:
[4.10_tc-SS_fix_v1.1_(plus tc preheat)_SEE_WARNINGS.txt](https://github.com/TBXin/NFirmwareEditor/files/387444/4.10_tc-SS_fix_v1.1_.plus.tc.preheat._SEE_WARNINGS.txt)

RenegadeTitan, 2016-07-28:
@olokos please let me know if this works. i dont have rx200 so cant test this. just rename extension to .patch and put it in correct patches/ folder in nfirmware app. its for newest rx200 firmware 3.10(24-5-2016):
[3.10(24-5-2016)_tc-SS_fix_v1.1_(plus tc preheat)_SEE_WARNINGS.txt](https://github.com/TBXin/NFirmwareEditor/files/387450/3.10.24-5-2016._tc-SS_fix_v1.1_.plus.tc.preheat._SEE_WARNINGS.txt)

olokos, 2016-07-28:
Thank you very much. I'm not sure how to set base resistance, really. Do you mean locking the resistance after screwing in the SS coil?

I will build a new SS coil and let you know how it goes. Thanks again!

behead974, 2016-07-28:
Hi, would it be possible to make a patch to activate Logo for RX200S? there is a patch available for the spash2 RX200 but not for S. Thank you

RenegadeTitan, 2016-07-28:
i just tested kanthal in tc mode with patch. used thin, tiny single coil for worst case scenario. i was expecting fireworks lol. it wasnt bad at all. 2 seconds of crackling juice. as long as cotton is damp then 2 seconds at 149.6w isnt bad. user can tell its firing too high and then it switches (correctly) to vw mode. setting wrong base resistance of tc coil will result in even less of a problem than kanthal. not firing kanthal in tc mode and correctly setting coil resistance should be common sense so u can take out all the warnings from description except the battery one (which should be common sense too but involves math so not everyone will figure that one out lol) if you think they are too alarmist.

@olokos yeah exactly. just locking in correct ohm when coil is at room temp. your welcome. 

@behead974 i may be only one working on joyetech chip patches right now so i will look into converting original patch. i will leave credit to original patch creator.

RenegadeTitan, 2016-07-28:
@maelstrom2001 i could use some tips if you remember what disassemble settings you used. im using ida arm little endian with default settings. i think im not setting read only/rom/ram correctly for c decompiling. doin ok figuring out code but i think its converting variables to constants because of my settings. the better i can see the code the faster i can understand it. any help will be appreciated. im good at figuring stuff out and have extensive tech experience but new to decompiling/disassembling.

maelstrom2001, 2016-07-28:
@RenegadeTitan I don't use C plugins in IDA, all modifications was done in assembler. And yes, all settings by default are OK for decompiling.

RenegadeTitan, 2016-07-28:
@maelstrom2001 wow. those are some really complex patches for being done all in assembler. ill have to learn more about these chip instructions as displayed in assembler. guess i have a lot of reading to do lol. thank you for your info and quick reply.

RenegadeTitan, 2016-07-28:
@maelstrom2001 can you explain what programs/apk/etc i would need once i learn more about arm/thumb assembly language? do i need to use an apk? knowing the process you used will help me narrow down what i need to learn.

maelstrom2001, 2016-07-28:
@RenegadeTitan I think I went the hard way, but at the same time it is also the simplest and most elegant solution IMHO. All you need is any ARM instruction evaluator, online or, for example, WAHC tool and branch calculator. Write patch in Assembler and convert it instruction by instruction into HEX code and place it into free/unused space in firmware, add necessary jumps to new code etc. I know it looks complicated and inefficient, but in fact it doesn't take a lot of time :)

RenegadeTitan, 2016-07-28:
@maelstrom2001 thank you! thats exactly the info i needed. it doesnt sound like the hard way (unless there is some easier way im not thinking of lol) thats roughly what i was expecting :) as i know more i may have more questions but now i know what i need to learn.

TBXin, 2016-07-29:
@RenegadeTitan max rom size is: 128kb (theoretically), so, knowing that you can do two things to get enough space for your patches:
1. Use some space inside firmware, i mean use same space as some unused glyphs does.
2. Take some extra space outside original firmware. Just enlarge firmware file up to needed size and put your patch body there.

ReservedField, 2016-07-29:
Throwing a couple cents in here on using extra space for patches.
ROM is indeed 128kB (RAM is 32kB in case you're wondering), but there are 4 2kB dataflash pages at the end, so APROM space is actually 120kB.
Also, logo in is the 2kB page starting at 0x19000, so 100kB is the actual limit.
You may have to [jump through some hoops](https://www.reddit.com/r/electronic_cigarette/comments/3zl349/tutorial_evic_vtc_power_to_5_letters_of_your/cz1n4w7) to reach your patch code because of the limited addressing space provided by 2-byte Thumb instructions (but it's a Cortex-M4 so you might be able to hack Thumb-2 32bit instructions in there, or you might just get lucky with indirect adressing).

maelstrom2001, 2016-07-30:
@RenegadeTitan let's write patch together :)
There is request for display brightness control in RX200S, so...
Open firmware in IDA and NFE. Find in IDA display init sequence by searching 0xAF - this is ON command for display controller. We have to add before it command 0x81 followed brightness value. As RX200S can be assembled with 2 types of displays, there are 2 separated subs:

[image]

[image]

Find in NFE unused glyphs:

[image]

There is no reference to them in code, so we can use it for additional commands, address and size you can find in NFE also, in status bar.
Glyph space will be used for some reloads of R0 an calls sub_49D4, which sends byte stored in R0 to display controller:

[image]

Replace 0xAF command in stock branches to 0x81 command, then replace BL sub_49D4 to B.W branch to our cut-in. It will be started with missed BL sub_49D4, then we will send brightness value and finally turn the screen on (0xAF as in the original). And B.W back :)

[image]

[image]

Upload modified firmware into RX200S and check. Finally, patch: https://dl.dropboxusercontent.com/u/43399149/VTC/tmp/RX200S4.10_Brightness100.patch

bluenazgul, 2016-07-30:
Since we need 0xF2 and 0xF3 to be Stock, would it be possible to create an
patch to "hide" this two images inside NFE?

<https://camo.githubusercontent.com/3c9cd49d3b8f0289fe07e39d3418acd48bee5585/687474703a2f2f692e696d6775722e636f6d2f4b536b6147524a2e706e67>

[image: Two]
<https://camo.githubusercontent.com/9c1be258960b405d05f25169173d520e309da7bb/687474703a2f2f692e696d6775722e636f6d2f693946787075582e706e67>

<https://camo.githubusercontent.com/eec0da8bc9df908bafda926f27bd386e6f0e56bf/687474703a2f2f692e696d6775722e636f6d2f67496e693646612e706e67>

<https://camo.githubusercontent.com/a8ba91e6ccf045c1fb02a5db4dcc48055ddc8b8e/687474703a2f2f692e696d6775722e636f6d2f4c59314b586e442e706e67>

<https://camo.githubusercontent.com/b242d588c81899f4bbffadfc5100cae4cbbb9146/687474703a2f2f692e696d6775722e636f6d2f467970645333742e706e67>

[image: Final2]
<https://camo.githubusercontent.com/45319fce28483d4da202797d210de805492ff9e6/687474703a2f2f692e696d6775722e636f6d2f334d53706b76572e706e67>

https://dl.dropboxusercontent.com/u/43399149/VTC/tmp/RX200S4.10_Brightness100.patch

TBXin, 2016-07-30:
@bluenazgul yes, it is possible. Selecting any glyph in the NFE you can see "reference" address in the statusbar. This reference is a absolute position in the firmware for this glyph in the image table. All that you need is navigate to this address and replace 4 bytes pointer to the 4 zero bytes (00 00 00 00).
After this manipulations NFE will not read this glyph anymore.

bluenazgul, 2016-08-01:
@TBXin thanks, is there allready an info about used/unused glyph and wäre the used glyph are used? I would like to create a patch to hide all unused glyph and write a small guide now to Theme the "dummy" Firmware

TBXin, 2016-08-01:
@bluenazgul i don't think that that list exists. For each device and each fw version this list can be different.

bluenazgul, 2016-08-01:
@TBXin so modify, flashing and testing would be the only solution to find out what glyph are used for the RX200S in FW 4.10 - after an Update the "0x00" would be the same, i think

TBXin, 2016-08-01:
@bluenazgul obviously used glyphs can immediately exclude because you constantly see them on the screen. 
All others can me determined by two ways: 
first - as you say, modify, upload, look.
second - disassemble firmware, look for glyph usage in the code.

bluenazgul, 2016-08-01:
@TBXin would it possible, to add an "write protection" for special glyph with an patch? So you could See the image, but cant edit it?

TBXin, 2016-08-01:
@bluenazgul without special support from the NFE - no, but it is appears some sort of surrogate.

MoCityTX420, 2016-08-01:
@TBXin When will this be available for download on mac products?

TBXin, 2016-08-01:
@MoCityTX420 no. You can use Wine or any kind of virtual machine to run this software.

MoCityTX420, 2016-08-01:
Where do I get Wine, or what other virtual machine would you suggest and where do I get them?

MoCityTX420, 2016-08-01:
@TBXin Where do I get Wine, or what other virtual machine would you suggest and where do I get them?

TBXin, 2016-08-01:
@MoCityTX420 VirtualBox. Just google it. Where you will get Microsoft Windows image - i don't know.

RenegadeTitan, 2016-08-02:
thank you guys for your super detailed explanations. totally get how make room for new code and jump/branch it. the explanation should be helpful to whoever sees this. the code example was terrific but brought up tons of questions regarding understanding the code (im learning arm assembly). im trying to avoid asking questions by reading thru arm manuals and have answered most the questions myself but i still have a few. how do we know that 0xAF is instruction for screen on? it looks like these values(registers)(im still learning the terminology) are sent to memory addresses 0x20000124 and 0x20000032 in sub 49d4. am i understanding the code correct? are these addresses and values set by manufacturer or in firmware? is nuvoton actual manufacturer?

maelstrom2001, 2016-08-02:
> how do we know that 0xAF is instruction for screen on?

Displays used in RX200S (as in VTC Mini, Cuboid etc.) are based on SSD1306 and SSD1327 controllers, all necessary information can be found in their datasheets.

> are these addresses and values set by manufacturer or in firmware? is nuvoton actual manufacturer?

In fact, in some cases not necessary to understand the structure of stock routines, as in our example. Sometimes that is enough to use everything that exists, especially in cases where there is nothing to do with hardware directly. Yes, more information you can get from Nuvoton datasheets for M451 series microcontrollers. And of course, from evic-sdk by ReservedField.

bluenazgul, 2016-08-05:
@TBXin  @maelstrom2001 i made an pull request with an "hide unused" patch for the RX200S FW4.10

bluenazgul, 2016-08-05:
@maelstrom2001 @RenegadeTitan would it be possible for you, to port the "Brightness Patch" and the "1W Steps Patch" from the RX200S (FW4.10) to the RX2/3 (FW4.12) without own the Device?

RenegadeTitan, 2016-08-13:
@bluenazgul i traded out my rx200s for a rx2/3 so i ported the patches i currently use. for now you can manually make the changes by using NFirmwareEditor function: encrypt/decrypt without opening the firmware. convert original to decrypted. then use a hex editor to change byte pairs as listed. then use NFirmwareEditor same function(encrypt/decrypt) to save unencrypted now changed firmware as encrypted.

RX2/3 FW4.12:

tc_SS_fix:
354A: C8 - 1C
354B: 71 - 61

1W_increments:
(credit to: maelstrom2001)
4032: 69 - 00
414C: 69 - 00
4288: 69 - 00
43F6: 69 - 00

i am currently using both so i can confirm they work. ive been busy but if i can ill try to catch up with a couple previous requests when possible.

olokos, 2016-08-21:
@RenegadeTitan sorry about the delay, I was quite busy and I just built my ss coil yesterday. As soon as it starts switching to VW I'll try the patch although it's quite possible that this problem happens only when your coil is not screwed in hard enough, well, we'll see. I'll be following this topic and as soon as it switches to VW I'll apply your patch.

RenegadeTitan, 2016-08-22:
@olokos the problem is caused by a 30-40 watt coil check. if 40w isnt enought to heat coil significantly within 2 sec or so it switches to VW. so VW mode switch only affects thicker coils that need more than 40w to heat up. the patch does however remove the 40w cap on coil check so any tc coil set higher than 40w will no longer start firing at 40w before switching to user set wattage. it will slightly improve vape on any tc coil set above 40w. the more so the higher the wattage needed/used for the coil. side note: if there is any loseness anywhere (coil, 5/10, etc) you may notice bigger wattage ranges as mod adjusts to maintain max temp. now that i wrench down my attomizer (not with an actual wrench lol) it tends to jump around within a 3 watt range with steady air flow. much more accurate than i thought mod was capable of. also, based on displayed temperatures and ohm difference between them, i believe mod (rx200s and rx2/3 so prob all of them with same chip) are actually reading accuracy at .001 ohms durring TC mode despite specs and only displaying .01 ohms. i would love to know if i am right about that.

bluenazgul, 2016-08-22:
Wismec just released FW 4.02 (200W) and 4.12 (250W) with offical Logo Support

https://github.com/TBXin/NFirmwareEditor/issues/64

Could you guys please try to port the current patches to the new versions?

RenegadeTitan, 2016-08-24:
@TBXin please add:
[RX23_4.12_TC_SS_fix.txt](https://github.com/TBXin/NFirmwareEditor/files/434100/RX23_4.12_TC_SS_fix.txt)

TBXin, 2016-08-24:
@RenegadeTitan ty for the patch. I add it to the repository.
[image]

wastez, 2016-08-26:
Could somebody port the patches for the RX200s to the new firmware? (4.12)

wastez, 2016-08-28:
I looked for some patches by myself.

Can somebody check if this is correct?
TC SS fix:    
345A: C8 - 1C
345B: 71 - 61

Voltage Cutoff 3V:
386A: 9B - 96

RenegadeTitan, 2016-08-28:
@wastez those locations/values for tc ss fix on 200s new fw4.12 are correct (dont have 200s any more and dont like to 100% confirm without testing)

wastez, 2016-08-28:
@RenegadeTitan
I'm also pretty sure that the tc ss fix. 
What do you think about the cutoff?

wastez, 2016-08-28:
Is there a Secure way to reflash it?
I was reading something about it but never tested it.

RenegadeTitan, 2016-08-28:
@wastez just checked voltage cutoff and yes that appears correct for 3.0v cutoff. there is a note in russian in the original patch for a 2nd value change that refers to "flashing batteries". i assume its the low battery indication. that value appears to be the same location in FW4.12 as FW4.10 (6b2: 9b - 96) but you may want to leave that value at the default 3.1v so you get an early low battery warning. however original firmware has these 2 locations set at the same voltage (3.1v) so that appears to work either way. this 2nd value is just a note in original patch and by default does not make the 2nd change. just wanted to make you aware of the option and that it is same location/value for both fw.

RenegadeTitan, 2016-08-28:
@wastez the removing batteries and holding down right button as you connect usb appears to flash the mod the same way in my experience. the reason it is most useful is that if your mod is bricked it should allow your computer to still recognize your mod and flash a new working FW to it (i may have some of the details of this incorrect but i believe my general point is correct)

wastez, 2016-08-28:
Ok thanks for the informations. 
I hex edited a lot in my life but i'm new in this sector, so thanks for your help.

RenegadeTitan, 2016-08-28:
same here :) still studying these firmwares like the dead sea scrolls lol. your welcome

wastez, 2016-08-28:
Just a question. 
Why did the russian guy set the flashing battery inidicator to 4.10V?
This doesn't make no sense if it was set to 3.10V before.

RenegadeTitan, 2016-08-28:
yeah i noticed that. its just a note tho so isnt an active value change by default. doesnt actually make changes unless you remove the # so im assuming he figured it didnt matter.

TBXin, 2016-08-28:
@RenegadeTitan 4.1 value for the test to confirm that the patch works as it should.

wastez, 2016-08-28:
I also thought about it that he maybe tested with this value. 
So the battery inidicator value should inidicate the 0% mark i guess. Correct?

olokos, 2016-09-11:
@RenegadeTitan 
https://github.com/TBXin/NFirmwareEditor/issues/49#issuecomment-235764759
This patch FOKIN FIXED SS! You are my lord and savior. I've been on Vape Expo 2016 Poland and the owner of HandVape.eu - Slava Chrome built a coil for me on the bellus, so there could be nothing wrong with the coil itself. Even though a master built it, I was still getting switches to wattage and atomizer short. He suggested a broken oring could be the cause. I replaced the oring and yeah, no atomizer short anymore, but it still switched to wattage by itself on ss. You just fixed it for me. Thanks mate. Could you tell me how do I port it to different firmwares? (Just one behind so that I could use this patch with the other patches)

wastez, 2016-09-12:
@olokos
Should be simple for the ss fix. 
Just decrypt the Firmware and open it with a hex editor. 
Search for C871 (should Beitrag just one entry in it, thats the reason why it is simple) and replace it with 1C61. 
The encrypt it and voila you have a patched Firmware.

[41 more replies]

closed 2016-10-02 by TBXin