TBXin/NFirmwareEditor #50: Brightness patch for RX200s

Ramjet428, 2016-07-21:
Could you do a 75% and 100% Brightness patch for the RX200s, this would make it a lot easier to read for those of us with poor eyes. Thanks.

Ramjet428, 2016-08-02:
For the brightness patch in the forum, you may want to add these description lines, as they tell what values can be used for various levels of brightness and how to turn the display positive or negative. I borrowed this from the Evic VTwo 4.02 brightness patch.

  <Description>brightness value 3F - 25%, 7F - 50%, C0 - 75%, FF - 100%</Description>
  <Description>pos/neg A6 (normal) / A7 (negative)</Description>

TBXin, 2016-08-06:
Patches were added to the NFE repository.

[image]

closed 2016-08-06 by TBXin