TBXin/NFirmwareEditor #52: 4.05 does not show patches for RX200 firmwares

olokos, 2016-07-23:
Hi,

I just downloaded NFirmwareEditor once again to check if any new patches came out, yet I was suprised to see none for 3.10 firmware downloaded from wismec website, so I tried to decrypt it, but nope. I can still edit all the images by hand, but no patches are ready.

Thanks :)

TBXin, 2016-07-24:
Hi,

You can use firmware storage: https://www.dropbox.com/sh/1438mnq977kpyss/AADeJFWBVmKx15YJkcuycuKRa
To get correct RX200 3.10 firmware file.

In your case you need Reuleaux_RX200_V3.10.bin, not Reuleaux_RX200_V3.10 (24.05.2016).bin (this version you downloaded from Wismec web site).

olokos, 2016-07-25:
I'm not quite sure why is this. Are the two firmwares any different apart from file name?

TBXin, 2016-07-25:
Wismec released two different firmwares with a same version.
Sometimes they do so...

olokos, 2016-07-25:
Thank you. Do you have any idea which one is better/newer?

TBXin, 2016-07-25:
RX200_V3.10 (24.05.2016) is newer than regular 3.10, but for end-user there are no visible changes during device use.

Also, there are no patches for 24.05.2016 version.

olokos, 2016-07-25:
Thank you very much :+1:

closed 2016-07-25 by olokos