TBXin/NFirmwareEditor #54: [Request] Patch to Remove puff timer Istick pico

TheHarlequin1, 2016-08-01:
Is there any chance to remove the puff timer from the istick pico? 
Would like to have the normal screen instead or maybe even show the applied voltage when pressing fire button. 

Besides it would be nice to shorten the screen on time to maybe 10 s. 

I don't know if it is even possible to patch the pico in this way. I would appreciate a response :) 

Thanks Steve

maelstrom2001, 2016-08-02:
Hello!
To the second part of your request, try this patches for setting screen dim timeout.
Firmware 1.00: https://dl.dropboxusercontent.com/u/43399149/VTC/tmp/pico/1.00_timeout10s.patch
Firmware 1.01: https://dl.dropboxusercontent.com/u/43399149/VTC/tmp/pico/1.01_timeout10s.patch

TheHarlequin1, 2016-08-02:
Thanks alot Maelstrom2001,

I could apply the patch for 1.01 without any problems. Screen now turns off after 10s.

I see so many usefull patches for the Evic (like showing battery Voltage, showing board temperature whilst charging via USB and so on...) but i have very little programming skills (mainly only what a mechanical engineer needs ;) ).

How is your workflow when creating patches? Do you decrypt the firmware and look where and how you have to make changes with a patch?

Thanks Steve

TBXin, 2016-08-02:
Thanks for the fast response.

Both patches were added to the NFE repository.

[image]

maelstrom2001, 2016-08-02:
@TheHarlequin1 thank you for response!
A little "guide" to creating patches you can find here: https://github.com/TBXin/NFirmwareEditor/issues/49#issuecomment-236360431
Unfortunately, more complex patches requires the presence a lot of the supported devices (for checking and debugging), so ATM we can provide a limited range of them. Thank you for understanding :)

closed 2016-08-04 by TBXin