TBXin/NFirmwareEditor #57: [Request] Patch RX2/3

bluenazgul, 2016-08-05:
Since we allready have the Firmware for the Wismec Reuleaux RX2/3
http://www.wismec.com/software/wismec-tc-software-for-upgrading-reuleaux-rx23/

Would it allready be possible to port the Patches (1W Steps / 100% Brightness) from the RX200S for the Rx2/3?

maelstrom2001, 2016-08-06:
@bluenazgul Please test:
https://dl.dropboxusercontent.com/u/43399149/VTC/tmp/RX23_4.12_brightness100.patch
https://dl.dropboxusercontent.com/u/43399149/VTC/tmp/RX23_4.12_1w.patch

bluenazgul, 2016-08-06:
Wow, thanks @maelstrom2001 both patches are working

Tested both seperate and together.

TBXin, 2016-08-06:
Both patches were added to the NFE repository:

[image]

closed 2016-08-06 by TBXin