TBXin/NFirmwareEditor #59: [REQUEST] Can you add various patches for egrip 2  v4.03

deddy84, 2016-08-11:
Hi, please can you add patches for joyetech egrip 2. I found another patches in nfirmwareeditor folder, but dont work whit egrip2. Thank you!

TBXin, 2016-08-11:
There are no patches for eGrip II firmwares.

deddy84, 2016-08-11:
I know! I found egrip 2 folder, but there is no patches in it. Can you add some patches in next release?

TBXin, 2016-08-11:
Patches is not my prerogative, also no one of us does not have eGrip device to test any modifications.
Maybe you know what patches do you need? :)

SaschaK, 2016-08-11:
There are some folders without content. They are placeholders. If patches exists they took them in. But no patches exist.

deddy84, 2016-08-11:
Need: percentage battery, max display brightness, 1 degree increase/decrease in tc mode, digital clock in screensaver. 
Of course, i can test patches and report to you. 
(sorry for my english)

closed 2016-08-22 by deddy84