TBXin/NFirmwareEditor #61: Reulaux 2/3?

SnoopDank, 2016-08-15:
Could you update your program to support the 2/3? It should be a simple patch.

TBXin, 2016-08-16:
Just add to your definitions.xml following

    
    <FirmwareDefinition Name="RX23 4.02">
    	<Marker Offset="0x90" Bytes="0x15 0x23" />
    	<Marker Offset="0xC0" Bytes="0x29 0x72" />
    	<Marker Offset="0xA30" Bytes="0x40 0xF2" />
    	<ImageTable1>
    		<OffsetFrom>0xC724</OffsetFrom>
    		<OffsetTo>0xCB24</OffsetTo>
    	</ImageTable1>
    	<ImageTable2>
    		<OffsetFrom>0xA894</OffsetFrom>
    		<OffsetTo>0xAC94</OffsetTo>
    	</ImageTable2>
    	<StringTable1 TwoByteChars="true">
    		<OffsetFrom>0xCC28</OffsetFrom>
    		<OffsetTo>0xCE7D</OffsetTo>
    	</StringTable1>
    	<CharsToCorrect>0x88 0x91 0x9A</CharsToCorrect>
    </FirmwareDefinition>
    
    <FirmwareDefinition Name="RX23 4.12">
    	<Marker Offset="0x90" Bytes="0x15 0x23" />
    	<Marker Offset="0xC0" Bytes="0x29 0x72" />
    	<Marker Offset="0xA30" Bytes="0x4F 0xF4" />
    	<ImageTable1>
    		<OffsetFrom>0xC724</OffsetFrom>
    		<OffsetTo>0xCB24</OffsetTo>
    	</ImageTable1>
    	<ImageTable2>
    		<OffsetFrom>0xA894</OffsetFrom>
    		<OffsetTo>0xAC94</OffsetTo>
    	</ImageTable2>
    	<StringTable1 TwoByteChars="true">
    		<OffsetFrom>0xCC28</OffsetFrom>
    		<OffsetTo>0xCE7D</OffsetTo>
    	</StringTable1>
    	<CharsToCorrect>0x88 0x91 0x9A</CharsToCorrect>
    </FirmwareDefinition>

TBXin, 2016-08-16:
Included in the 4.06 release
https://github.com/TBXin/NFirmwareEditor/releases/tag/4.06

closed 2016-08-16 by TBXin