TBXin/NFirmwareEditor #62: Joyetech modified the 3.03 image

ClockSelect, 2016-08-17:
Almost everything is in the title^^
This is a usual habit of Joyetech to modify an image version without telling anyone and keeping the very same version number.
NFE uses a 3.03 version dated 31.05.16, new 3.03 is June.
Obsiously, resource database does not match.
A new definition may be necessary.

ClockSelect, 2016-08-17:
Modifs for the new 3.03: (old in comments)

	
	<FirmwareDefinition Name="Evic VTC Mini 3.03 (31.05.2016)">
		<Marker Offset="0x90" Bytes="0x35 0x23" />
		<Marker Offset="0xC0" Bytes="0x2D 0x7A" />
		<ImageTable1>
			<OffsetFrom>0xD324</OffsetFrom>
			<OffsetTo>0xD724</OffsetTo>
		</ImageTable1>
		<ImageTable2>
			<OffsetFrom>0xB494</OffsetFrom>
			<OffsetTo>0xB894</OffsetTo>
		</ImageTable2>
		<StringTable1 TwoByteChars="true">
			
			<OffsetFrom>0xD828</OffsetFrom>
			<OffsetTo>0xDA8B</OffsetTo>
		</StringTable1>
		
		<CharsToCorrect>0x88 0x91 0x92 0x9A</CharsToCorrect>
	</FirmwareDefinition>

ClockSelect, 2016-08-17:
Markers will not discriminate, though.

closed 2016-08-17 by ClockSelect

TBXin, 2016-08-17:
@ClockSelect where did you get June version of the 3.03 firmware?
I've just downloaded UpdateFirmware.zip from Joyetech website for eVic VTC Mini 3.03 and there 31.05.2016 version.

TBXin, 2016-08-17:
Also, definition that you given identical to the one that is in the repository, except 0x92 glyph in the CharsToCorrect attribute.

ClockSelect, 2016-08-18:
They reverted to an older version.
0x92 is the "j", no matter (they forgot it)
See the version my repo is based on, it's june revision. You're right, it doesn't exist any more; they reverted to May revision.