TBXin/NFirmwareEditor #64: RX200S FW4.02 // FW4.12 released

bluenazgul, 2016-08-22:
Wismec just released FW4.02 (200W) and FW4.12 (250W) for the RX200S with official Logo Support

http://www.wismec.com/software

Would it possible to add these two versions to NFE?

And for the Patch creators please port the RX200S patches to the new Firmware (maybe the RX2/3 Patches could help here)

TBXin, 2016-08-22:
Update your definitions.xml with following:

    <FirmwareDefinition Name="RX200S 4.02">
     <Marker Offset="0x90" Bytes="0x25 0x22" />
     <Marker Offset="0xC0" Bytes="0x29 0x71" />
     <Marker Offset="0x1A64" Bytes="0xC9 0x70" />
     <ImageTable1>
      <OffsetFrom>0xC624</OffsetFrom>
      <OffsetTo>0xCA24</OffsetTo>
     </ImageTable1>
     <ImageTable2>
      <OffsetFrom>0xA794</OffsetFrom>
      <OffsetTo>0xAB94</OffsetTo>
     </ImageTable2>
     <StringTable1 TwoByteChars="true">
      <OffsetFrom>0xCB28</OffsetFrom>
      <OffsetTo>0xCD7C</OffsetTo>
     </StringTable1>
     <CharsToCorrect>0x88 0x91 0x9A</CharsToCorrect>
    </FirmwareDefinition>
    
    <FirmwareDefinition Name="RX200S 4.12">
     <Marker Offset="0x90" Bytes="0x25 0x22" />
     <Marker Offset="0xC0" Bytes="0x29 0x71" />
     <Marker Offset="0x1A64" Bytes="0xCE 0x70" />
     <ImageTable1>
      <OffsetFrom>0xC624</OffsetFrom>
      <OffsetTo>0xCA24</OffsetTo>
     </ImageTable1>
     <ImageTable2>
      <OffsetFrom>0xA794</OffsetFrom>
      <OffsetTo>0xAB94</OffsetTo>
     </ImageTable2>
     <StringTable1 TwoByteChars="true">
      <OffsetFrom>0xCB28</OffsetFrom>
      <OffsetTo>0xCD7C</OffsetTo>
     </StringTable1>
     <CharsToCorrect>0x88 0x91 0x9A</CharsToCorrect>
    </FirmwareDefinition>

Magical8fingers, 2016-08-28:
TBXin, sorry this might sound dumb, but how do I do the above???........

TBXin, 2016-08-28:
@Magical8fingers you need to open Definitions.xml file that is located near NFirmwareEditor.exe and put text that i provided above like other same nodes in the file.

Magical8fingers, 2016-08-28:
Sorry to be a pain, could you give me a more in depth how to please!...

TBXin, 2016-08-28:
@Magical8fingers 
Just replace your file with file from this archive...
[Definitions.zip](https://github.com/TBXin/NFirmwareEditor/files/441417/Definitions.zip)

Magical8fingers, 2016-08-28:
[image]
Sorry I'm a noob to this lark ;) lol

Thank you kindly

closed 2016-09-07 by TBXin