TBXin/NFirmwareEditor #65: VTwo Mini Dclock.patch problem

mirko93s, 2016-08-24:
Clock is too high and it covers partially the analog clock.
Fw 4.02

[image]

TBXin, 2016-08-24:
Hello,

1. You can redraw gplyph 0x107;
2. You can fix patch for yourself;

If you choose second way, first of all you need to rollback this patch, then open patch file via text editor of your choice, look at this lines and change right part to desired Y position (hex value) for each time part (hours - y1, minutes - y2, seconds - y3). After this manipulations you can apply fixed by you patch.

```
D050: 00 - 04 ; y_1
D05A: 1F - 04 ; y_2
D07E: 14 - 04 ; y_3
```

mirko93s, 2016-08-24:
I try to fix hex value but I prefer to clear glyph 0x107 and have a clean UI.
I received an update about this patch but it doesn't seem to work.
Thanks a lot anyway

TBXin, 2016-08-24:
> I received an update about this patch but it doesn't seem to work.

That's why i recommended to create a copy of the patch. NFE checks that all patches are same in your local storage and in the remote repository.

Unfortunately i can't test myself because i dont have VTwo Mini.
I can fix patch in the repo if you provide correct values for y_1, y_2 and y_3.

closed 2016-09-29 by TBXin