TBXin/NFirmwareEditor #67: Bricked Rx200 :OO?!?

Kombohead, 2016-08-26:
Hi since i updatet the secons ( s when firing ) into this flame icon and updating the rx200s it crasehd and is now black the update tool also crashes....... what can i do 
[image]

Kombohead, 2016-08-26:
fixed it self sry

TBXin, 2016-08-26:
Hello,

You can upload original or custom working firmware when device is in the safe mode.

1. Take out batteries;
2. Press + button and hold;
3. Connect to the USB;
4. Release + button;
5. Open NFE Flasher or Wismec ;
6. Device will recognized with Firmware version: 0.00;
7. Upload firmware;
8. Profit;

closed 2016-08-26 by TBXin

Kombohead, 2016-08-26:
thanx!!!!
can i also somehow change the layout of the screen ?

TBXin, 2016-08-26:
There are no simple way to do this.
Only way is modifing executable code of the firmware. This is a complex task.

Kombohead, 2016-08-26:
ok so i cant e.g insert a fullsice logo instead of loading symbol ?

TBXin, 2016-08-26:
You can safely replace any glyph with your own image without changing glyph size.
Also, you can decrease original glyph size, but you can't increase it size without creating patch, that will relocate original glyph to the other space in the firmware.

Kombohead, 2016-08-26:
ok ty so no bigger Logos for Noah :D (me)