TBXin/NFirmwareEditor #69: Compile NFE from Source

bluenazgul, 2016-08-30:
Do I need anything else to Visual Studio 2015? To get NFE build with downloaded Github Source?

TBXin, 2016-08-30:
Nope, only VS 2015 to build project from sources.

Bidulglloq, 2016-09-04:
Might it work from mac, e.g. installing VS2015 for mac and compiling the sources?
-or does your source code contains MSwindows specific code, preventing porting it to Mac OS?

TBXin, 2016-09-04:
@Bidulglloq 
1. There are no any VS for mac
2. Yes, UI is all windows-specific, other code can be ported.

Bidulglloq, 2016-09-04:
Sad for that...
Any plans for a mac version? ;)

TBXin, 2016-09-05:
@Bidulglloq unfortunately no, but you can try to run NFE with WINE, also you can use Parallels / VirtualBox to install windows and use NFE.

closed 2016-09-07 by TBXin