TBXin/NFirmwareEditor #72: myevic

ClockSelect, 2016-09-06:
About:
[myevic](https://github.com/ClockSelect/myevic)

After system vectors, at offset 0x140, myevic firware will contain:

0x140: Pointer to start of picture table block 1
0x144: Pointer to end of picture table block 1
0x148: Pointer to start of picture table block 2
0x14C: Pointer to end of picture table block 2
0x150: Pointer to first string
0x154: Pointer to end of last string+1

Would it help make it editable?

TBXin, 2016-09-06:
Hello,

Thanks for the pointers. This is cool feature, that allowing to use only one definition for every version of myEvic firmwares.

I will add support to the NFE soon.

ClockSelect, 2016-09-07:
Great, thanks :)

TBXin, 2016-09-07:
@ClockSelect released NFE 4.1 with your firmwares support.
https://github.com/TBXin/NFirmwareEditor/releases/tag/4.1

closed 2016-09-07 by TBXin

ClockSelect, 2016-09-08:
Awsome!
Grats!