TBXin/NFirmwareEditor #74: [Request] Li-FePo4(A123) support for RX2/3

Arrimf, 2016-09-08:


TBXin, 2016-09-08:
For what? PCB does not allow to take more than 25 amps.

bluenazgul, 2016-09-08:
Thats not correct @ReikoKitsume - Wismec says, RX2/3 and RX200S have an
block at 50 amps - max tested by myself was 47,3 amps shown on LCD

Arrimf, 2016-09-08:
For they(akks) shows better behavior on negative temp.
+ longer lifetime (>1000 cicles ¦¦ ~3 years)
+ opportunity..))

RenegadeTitan, 2016-09-15:
TBXin said "takes" as in input from batteries. not output from 510(which is 50a). however the firmware will work with 2 26650s if u can get x2 26650s to touch same hardware contacts as x2 18650s. because of 25a limit (safely assuming TBXin is correct) 26650s wont add wattage but will match 18650 voltage range so should work fine without software change. hardware change however... gotta make em fit. im guessing u meant 1 26650. i believe it could be done by modifying extra code in 2/3 4.12 thats not in 200s 4.12 since thats where 2 or 3 batteries check is and that code already bypasses 1 of the 2 battery voltage check wires. HOWEVER because your voltage to chip is now 3.0-4.2v and not 6.0-8.4v your max output wattages will basically be cut in half. theres a reason max output wattage of 3 batteries is 250w and 2 batteries max output wattage drops to 200w. max input from 2 batteries: 8.4v*25a=210w (minus chip efficiency loss) so 1 battery would be less than 105w at full charge of 4.2v and less than 75w as battery approaches dead 3.0v charge.

closed 2016-09-29 by TBXin