TBXin/NFirmwareEditor #75: [REQUEST] definizione for 4.04 evic aio

deddy84, 2016-09-08:
Please, can you add support to 4.04 evic aio?

TBXin, 2016-09-08:
Already done.

deddy84, 2016-09-08:
Thanks, for 5.04 definition
but i want to open 4.04 evic aio firmware. nfirmwareeditor say: unable to load firmware. No one definition is not appropriate for the selected  firmware file.

TBXin, 2016-09-08:
Oh, i see.

TBXin, 2016-09-08:
Restart NFE to get eVic AIO 4.04 support.

[image]

closed 2016-09-08 by TBXin

deddy84, 2016-09-08:
2 words: THANK YOU!

deddy84, 2016-09-08:
Don't find big clock screensaver, so i'm unable to import from my old custom firmware.

TBXin, 2016-09-08:
[image]

deddy84, 2016-09-08:
Sorry, but that is the small clock in power mode. 
Please check 4.02: at the bottom of the list was another bigger watch, which was the screen saver.

TBXin, 2016-09-08:
Can you provide screenshot with big clock? I dont see them in the firmware.
Also, at the screenshot that i provided - opened 4.02 and 4.04 firmwares and there are no big clocks...

TBXin, 2016-09-08:
Oh. It's my bad. A little mistake in the firmware reading code. I wll create new NFE release soon.

deddy84, 2016-09-08:
A BIG THANKS AGAIN! :)

TBXin, 2016-09-08:
Done.
[image]

[image]