TBXin/NFirmwareEditor #80: Same patch in evic aio

NasosGR, 2016-09-15:
Hello there, I have an Evic Aio and i would like that patch on that kit. Now i dont have a clue about proigramming and it took me long enought to get here. I found firmware editor and saw I can aply patches to firmwares. So now I am looking how can I add that patch on my Evic Aio Software. Please help me. My email is N.tsimpliaridis@gmail.com

TBXin, 2016-09-15:
Hello,

There are no patches available for eVic AIO firmwares. 
No plans to porting patches. So, your only hope wait when someone will create patches for your device firmwares.

closed 2016-09-15 by TBXin

NasosGR, 2016-09-15:
But the firmwares are more or less all the same, isnt there anyway I can import the patch to the evic aio firmware?

TBXin, 2016-09-15:
This is not true. You can make such a conclusion only as a user, but at the binary level each firmware is almost unique.
Thats why anyone can not create patches without disassembling firmwares to the ASM.

NasosGR, 2016-09-15:
I understand... Look I am sorry if I am wasting your time but I am really determined to make this work, is there any way I can contact you via email or something?

TBXin, 2016-09-15:
Unfortunately i can not help you with patches. I can recommended to you read this: https://github.com/TBXin/NFirmwareEditor/issues/49
There are some little guide how to port patches.