TBXin/NFirmwareEditor #81: Trojan: Win32/Fethar.B!cl found on version 4.11

StormtrooperBF, 2016-09-16:
I really do like your work.

But after windows defender found your trojan, I cannot open the editor.
So please delete it.

StormtrooperBF, 2016-09-16:
[image]

TBXin, 2016-09-16:
Oh, just lol.

What should i delete? Your windows defender?
Did you know about false-alarm? So, if you dont - this is it.

If you do not trust binaries that i provided you can comple NFE from sources.

Do not trust me? Use normal anti-virus software, not windows defender.
There are many cool services that can check files for viruses. like that: https://virustotal.com/en/file/730fe208e64126a2307817aed3a2dea79aed06e8340fdc0a17224aa3e7593ecb/analysis/1474023825/

closed 2016-09-16 by TBXin