TBXin/NFirmwareEditor #86: Export/import mod settings

Dareksa, 2016-09-23:
Is it possible to add export/import mod settings via NFE Edition-Configuration ?

TBXin, 2016-09-29:
Hello,

Yes, it possible, but what kind of settings do you want to export/import?
If you don't want to share settings with other people you can save the dataflash via Firmware Updater.
Dataflash contains all settings and other system stuff like HWVer.
So, if you save dataflash from your device later you can safely wirte this dataflash back to your device.

Dareksa, 2016-09-29:
Hello,

I had in mind export settings from one device to another, for example, from VTC mini to vtwo mini.
I use both models in parallel and at the moment I have to set them up separately.

TBXin, 2016-09-29:
Okay, i understood you.
Will see what i can do.

closed 2016-12-17 by TBXin