TBXin/NFirmwareEditor #93: A couple istick pico patches for 1.00

joeblowma, 2016-10-15:
Not really an issue... just hoping to be helpful to others. I'm finding the eleaf istick pico 1.00 firmware to be more useful than the later versions due to the voltage display in VW mode, with some minor annoyances... which I've attempted to correct here (without hopefully learning how to do a pull request :o )

I've modified the already existing screen timeout patch for 1.00 to do 5s instead of 10s, may or may not want to include it (credit to the original author intact, thanks NFirmwareEditor Team!) as really its not a difficult change for anyone that wants to do it.

I've also written a new patch for 1.00 based loosely on the button mod patch for 1.01, that makes the coil menu only accessible from power off via 5 clicks, makes the power on/off 3 clicks instead of 5 (found myself accidentally hitting it at 2 from time to time), makes the temp display at 6 clicks (instead of 10) from power off, and fw version display at 7 (instead of 20) from power off.

Both patches are in the zip attached
[iStickPico1.00_patches.zip](https://github.com/TBXin/NFirmwareEditor/files/531495/iStickPico1.00_patches.zip)
Take it or leave it, I've only been able to test on my own singular device and they both work exactly as described. Also, I don't see any conflicts with the other specified buttons/combos in the manual, but I only use this device in VW mode. I can do further testing/limited modifications if needed as well, and though I'm good at ARM finding where to add/modify a longer timeout for the puff 'meter' currently eludes me.

And finally, if nothing else, thanks for the great NFE software, everyone who has contributed! Works like a charm!

TBXin, 2016-10-18:
Hello,

Thanks for your work.
Just added patches to the NFE rep. https://github.com/TBXin/NFirmwareEditor/commit/00367fd453b51251881a615e11c3882b4da48128

closed 2016-10-18 by TBXin

isaso, 2017-01-12:
Hi joeblowma,

Quick question, I want to try the screen timeout patch for 1.00 to do 5s. 

Does this work for V1.03 firmware? If not, is there any way that I can change it?

Regards.

TBXin, 2017-01-12:
@isaso i'd recommend to you use this firmware, instead of original.

https://github.com/maelstrom2001/ArcticFox

isaso, 2017-01-13:
Thank yous so much!

joeblowma, 2017-01-13:
Ah ha ha, I've been using Arctic Fox since they added pico and it is indeed great - no more need to mess with patches to change things how you like :D

I had ported all the patches to the latest official firmware, but was pointless because they forced their logo and did some strange things with button lockouts while logo showing that broke the button patches so I had gone back to 1.00... until AF :D

TBXin, 2017-01-13:
@joeblowma hehe))
FYI: I am one of the AF developers ^^

joeblowma, 2017-01-14:
@TBXin NFE without a doubt makes AF AAA, of course I know you are all in on that - honestly I think you'd see a grown man cry if forced to go back to official firmware. Plus, your user logo is a bit of a giveaway on that point too... ^^