TBXin/NFirmwareEditor #94: Wismec Presa TC 75W bricked using Editor and latest MyEvic NFE firmware

MaRCoMLXXV, 2016-10-19:
Not sure wether this is firmware or NFE editor related, but as I don't speak Russian I hope it's okay to post it here. I've been using your custom firmware as well as the Editor on one of my Presa TC 75W's for quite some time now without any problems whatsoever, but now I've ran into issues. 

**Issue Descriptions:**

When flashing the current build (161016a) it bricks my Wismec Presa TC 75W. Holding right button while connecting USB and uploading stock firmware is required to bring the mod back to live. NFE Editor reports that firmware upload succeeds but regardless which build is installed (stock firmware, ClockSelect MyEvic), it bricks the mod every time.

**Steps to Reproduce:**

Restore Presa TC 75W to stock and use NFE Editor to upload build 161016a to reproduce. Editor reports firmware upload succeeded, but device will never reconnect. Disconnecting USB and reinstalling battery shows that device is bricked. Issue did not occur with previous firmware.

Thanks in advance for your reply.

maelstrom2001, 2016-10-19:
Try to restore Presa TC 75W to stock firmware, then upload the latest build of myEvic (by ClockSelect or [this](https://dl.dropboxusercontent.com/u/43399149/VTC/myevic_fork_test/myevic_161019.bin)). If Presa won't start, remove battery, press minus button and install battery while holding minus button.
It seems the firmware bug, not NFE.

MaRCoMLXXV, 2016-10-19:
Thanks for the quick reply, I will try that. What does the procedure (minus button pressed while installing battery) do? Is it a soft reset?

TBXin, 2016-10-19:
Nope, recently this combinatios was added, and they should solves case when your device cannot start RTC module and hangs. When your are holding minus your are forcing to turn off RTC module on start.

closed 2016-10-25 by TBXin