TBXin/NFirmwareEditor #96: iStick pico mega vcom mode

xXxBobDolexXx, 2016-10-24:
When I plug my iStick pico mega into my computer and try to run the device monitor, it says compatible  device found but vcom connection could not be established, clicking ok to enable vcom does not do anything. Just wondering if theres something Im missing like a driver. 

Firmware 1.02 using eleafs own updater

TBXin, 2016-10-24:
Hello, 

If you look closely, you'll notice that the menu called "myEvic NFE Edition".
This is the name of the firmware, and therefore all the submenus just for that firmware.

Not 1.02 eleaf, not 3.14 joyetech, not anything else except myEvic NFE Edition.

There are no support from this firmware for devices like you have. Supported devices are:
eVic VTC Mini
eVic VTC Dual
eVic VTwo Mini
eVic VTwo
eVic AIO
eVic Basic
eGrip II
Cuboid
Cuboid Mini
Wismec Presa 75W
Wismec Reuleaux RX75
Wismec Reuleaux RX200S

xXxBobDolexXx, 2016-10-24:
Ah, makes sense. Thanks

closed 2016-10-24 by xXxBobDolexXx