TBXin/NFirmwareEditor #99: myEvic NFE Edition not recognizing my device

SEEKNDESTROY, 2016-11-02:
**Issue Descriptions: myEvic NFE Edition not recognizing my device
OS: Windows 10 (10.0.10586 Build 10586)
Mod = RX200S
Latest myEvic firmware installed
Expert Settings:
USB set to COM
NFE = On
UCH = Off

Steps to Reproduce:
1. Load latest myEvic firmware in NFirmwareEditor v.5.1
2. Plug in RX200S
3. Select "Device Configuration" - All I get is "_Waiting for device with myEvic NFE Edition 161002 or newer_"
4. Select "Device Monitor" says "_No compatible USB devices are connected. To continue, please connect one. If one already IS connected, try unplugging and plugging it back in. The cable may be loose._"
I follow the instructions, click OK and the same message. I click cancel and NFirmwareEditor stops working/crashes.

[image]
[image]
[image]

I hope this helps.**

TBXin, 2016-11-02:
Hello,

> Latest myEvic firmware installed

Waiting for device with myEvic **NFE Edition**

SEEKNDESTROY, 2016-11-02:
Hi TBXin,
Thank you for your quick reply.
My installed firmware is 161030 which should be newer than 161002, right?

TBXin, 2016-11-02:
Yep, 161030 is newer than 161002, but i mean that "myEvic" is not "myEvic NFE Edition". This is two different firmwares, and NFirmwareEditor (NFE) will work only with "myEvic NFE Edition"

SEEKNDESTROY, 2016-11-02:
Gotcha! Thanks for clarifying and sorry for the trouble.

closed 2016-11-02 by SEEKNDESTROY