TBXin/NFirmwareEditor #102: Device configuration causes weird exclamation mark

puyamn, 2016-11-06:
After updating the myevic NFE settings on my VTC Mini a weird exclamation mark appears on the right of the battery indicator, which I can not find in the images of the firmware version. Any help or fix?
[image] [image]

TBXin, 2016-11-06:
Hello, this is LSL Mode = On. You can turn off it in the Expert menu on the device.

puyamn, 2016-11-06:
I'm sorry, I just found the settings on device. Thanks for quick reply.

closed 2016-11-06 by puyamn