TBXin/NFirmwareEditor #106: VTC Mini - Expand 1st and 2nd line too

BigBadBlackWolf, 2016-11-21:
- The changes in the 3rd line for Vout and Batt as changeable select/move to the 1st line.
Selection 1st line: Volt -> Vout -> Batt

- The changes in the 3rd line for RES as changeable select/move the 2nd line.
Selection 2nd line: Coil -> RES

Selection 3rd line: Amp -> Puff -> Time -> Board(temperature) -> Clock (digital)

TBXin, 2016-11-21:
Hello,
Thanks for your suggestion.

Soon we will release our firmware in which you can change every line, even battery line using our configuration utility.

[image]

BigBadBlackWolf, 2016-11-22:
Thanks for the info TBXin!

As I can see so far from the picture, you can select the desired displayed value of each line (1 - 4) with the software itself. But probably always only one value and in addition Puff Time (when setting the checkbox). 
I thought however with my request / suggestion rather by the fact that one as before in the 3rd line several values ​​also for the 1st and 2nd line set and by further clicking in the selection menu of the VTC mini can select, which is indicated.
e.g.:
- Selection in the 1st line: VOLT -> VOUT -> BATT 
- Selection in the second line: COIL/(COIL Lock) -> RES 
- Selection in the 3rd line: AMP -> PUFF -> TIME -> BOARD (temperature) -> CLOCK (digital) 
as is currently possible in the custom firmware "myevic 161107" (only without my mentioned change regarding line 1 and 2).

closed 2016-12-17 by TBXin