TBXin/NFirmwareEditor #110: Arctic Fox and infos

forkoz, 2016-12-03:
So I see you have made your own firmware instead of myevic. I looked at the forum with translate but see no features or guide, only hints of cool things like making battery last longer, maybe log battery, etc.

Can you point me in the right direction? You say proprietary firmware.. is the source not available? I wanted to add voltage mode because I don't like power, it drains down the battery after coil resistance rises and sometimes I want to be under 3 volt on fresh coil. CS scoff at that idea because he thinks everyone vapes like him or will use it to fire tiny coils at full power.

Just got the v2 mini today so still learning.

TBXin, 2016-12-03:
Hello,

You can read about ArcticFox firmware here: http://www.ecigtalk.ru/forum/f16/t101098-p124.html#post9463684
Unfortunately at the moment descriptions available only in Russian.

> You say proprietary firmware.. is the source not available?

Hm, strange translataion from your online translator^^
Firmware is a part of the brand "NFE". Sources is not available at this moment. 

> I wanted to add voltage mode

I think that we can add this mode in future version. Need to discuss it with a colleague.

forkoz, 2016-12-03:
thanks for the quick response. I used google translate as on 4pda and all of the other sites I have to go to. many times its the only option. 

vv is like a unicorn when looking for devices. last one I found was cloupor mini plus and I've beat it up from daily use. It also wasn't PWM but step up/down. The clock and flash-ability of V2 won me over.

Logging only works on PC? It would be cool log discharge of the battery and rotate out weak ones. Maybe not enough memory?

TBXin, 2016-12-03:
Yeah, device can't store this kind of data, because rom is too small.

forkoz, 2016-12-03:
Wow, I just loaded with dropbox and set up with circle skin. There are a tiny bit of bugs (ie clock doesn't go away when adjusting in "classic", over protection happen on 2nd puff) but this FW is amazing. Props.

TBXin, 2016-12-03:
>  but this FW is amazing

Nice to hear.

> There are a tiny bit of bugs (ie clock doesn't go away when adjusting in "classic", over protection happen on 2nd puff)

Can you explain in more details?

forkoz, 2016-12-03:
I started with classic line based screen because I didn't know about the circle skin. When I hit + or - on watts the clock stays on the screen and I can't see voltage or resistance at that power. In the circle skin it works correctly.

10s protection seems like it goes too fast, I wish I could change this, if not with a setting maybe with patches + ida. So I stop puffing, release button to take another puff but protection shows on the screen instead of firing. When trying to reproduce it doesn't happen consistently.

TBXin, 2016-12-03:
Oh, i see.

1. This clocks behavior is not a bug. You can change 2-3-4 clicks actions in the menu -> settings -> clicks. e.g. set clock to the 2 clicks, and now you can change what you will see.

2. About 10s... it's a very strange. We never receive such issues.

forkoz, 2016-12-03:
maybe its the device? either way changing protection is good. obviously need some setting to prevent it blowing up in your pocket, 10, 15, 25 ,etc. If I used this with a wax pen it has to heat for a long time and 10 seconds way too short.

I also can't find the battery profile editor in configurator but I'll keep reading from the guide on.

TBXin, 2016-12-03:
> obviously need some setting to prevent it blowing up in your pocket, 10, 15, 25 ,etc

Turn off device by 5 clicks on fire button.

> I also can't find the battery profile editor in configurator 

Advanced -> Settings -> Battery Model -> Custom -> Edit button right after combo box.

forkoz, 2016-12-03:
I found the nightly and puff cut off at least up to 15s. Now its not tripping.

forkoz, 2016-12-03:
> Turn off device by 5 clicks on fire button.

True, I never do this, maybe I vape too much. 15s is really the perfect time for regular vaping or I got used to cloupor. For a wax pen I have to fire 20+ but I read the post where you say its too much stress on the board.

Finally found battery curve after that. Thanks!

closed 2016-12-18 by TBXin