TBXin/NFirmwareEditor #111: how to set maximum wattage on arcticfox?

mirko93s, 2016-12-05:
**Issue Descriptions:**

**Steps to Reproduce:**

TBXin, 2016-12-05:
Maximum wattage is limited by your device, e.g.  for VTC Mini is 75w, for cuboid is 200w and so on.

mirko93s, 2016-12-05:
Hi, thanks for the early answer, With patches I mod vtwo mini from 75 to 100w on ofw 4.02

TBXin, 2016-12-05:
The theoretical maximum for VTwo mini is 92.5 (3.7V * 25A) watts, and yes, this is only theoretical. In real life you device with fresh batteries can give to you this wattage a very little amount of time.
No one has repealed the laws of physics.

mirko93s, 2016-12-05:
So there's a way to increase max wattage? can I mod the bin if you kindly tell me what code line I need to see.
p.s. I translate the firmware in italian but some words are longer then english, can I add some more space to strings?

TBXin, 2016-12-05:
You can't increase max wattage in the ArcticFox firmware.
And also you can't increase length of strings.

mirko93s, 2016-12-05:
Ok, thanks anyway and congratulations for this amazing firmware. Sorry to bother you.

TBXin, 2016-12-05:
No problem! I glad to hear that you like our firmware.

closed 2016-12-05 by TBXin