TBXin/NFirmwareEditor #112: ArcticFox Configuration tool won't connect

cyris69, 2016-12-09:
**Issue Descriptions:**
ArcticFox Configuration tool does not connect to device. NFE Toolbox is 6.0
Here is a screenshot of all it does
http://i.imgur.com/AtHIkNW.png
The rest of the program works fine inlcuding live status
**Steps to Reproduce:**
For me running it with or without rx2/3 plugged in does it for me as I have never gotten that feature to work. Firmware is working fine on the device and is build 161205

TBXin, 2016-12-09:
Hello,

If you decide to use nightly builds of the ArcticFox Firmware, you also should use nightly builds of the NFE Toolbox.

Link to the nightly builds of the NFE Toolbox you can get at the main page of this repository.

cyris69, 2016-12-09:
My bad, I can't remember where I got the device build to be honest. I'll look here for the nightly tool build.

TBXin, 2016-12-09:
which builds you are using? toolbox + arcticfox

cyris69, 2016-12-11:
Sorry for the delay in response. NFirmwareEditor v6.0.34 since you mentioned I was on a nightly. Also, not sure if it's just a visual thing but when you lock the ohm it still changes on screen as the coil heats up.

TBXin, 2016-12-11:
COIL or RES ?

maelstrom2001, 2016-12-11:
@cyris69 Do you remember the behavior of COIL line in genuine Joyetech firmwares? Here is the same. While you pressing fire button, there is realtime resistance. Locking resistance means that mod will not ask you every time "is this new coil or same?" when resistance was changed slightly. But the appearance is close to stock firmwares: saved/locked Ohms in standby, and the real while heats up.

closed 2016-12-22 by TBXin