TBXin/NFirmwareEditor #113: Arctic Fox Compatibility?

MoDuLah, 2016-12-10:
Haven't seen any other issue like that or anywhere in the forum about device compatibility.
So on which devices can this be used (and has been tested on them)? 
Except VTC-Mini and Cuboid which i have tested it personally.
A list on that forum post or somewhere here would be nice to exist ;)

TBXin, 2016-12-10:
> A list on that forum post or somewhere here would be nice to exist ;)

Hm...

i see one ;)
[image]
and one more
[image]

We currently have no time to create ArcticFox landing page in few languages.
We're going to do it, but it is not clear when.

cvrle77, 2016-12-21:
Can you please include Presa 100w ? It doesn't seem included.

bluenazgul, 2016-12-21:
List of supported devices:

   - eVic VTC Mini
   - eVic VTC Dual
   - eVic VTwo Mini
   - eVic VTwo
   - eVic AIO
   - eVic Basic
   - eGrip II / Light
   - Cuboid
   - Cuboid Mini
   - Wismec Presa TC75W
   - Wismec Presa TC100W
   - Wismec Reuleaux RX75
   - Wismec Reuleaux RX200
   - Wismec Reuleaux RX200S
   - Wismec Reuleaux RX2/3
   - Eleaf iStick Pico 75W

TBXin, 2016-12-21:
Presa TC100W is supported. You can upload latest ArcticFox firmware to your box.

cvrle77, 2016-12-21:
Well, I just downloaded 6.2 version, and it says 6.1, and wants to update again to 6.2, which is 6.1,  and when I attach my device, ArticFox throws message: Unable to download device settings. Reconnect your device.

cvrle77, 2016-12-21:
And on the bottom, it says Device is connected. Ready.

TBXin, 2016-12-21:
For what reason you are launching NFirmwareEditor? :)
Upload latest ArcticFox firmware [161221] to your box via NToolbox -> Firmware Updater, than ArcticFox Configuration will work.
And yes, download again latest NFE release https://github.com/TBXin/NFirmwareEditor/releases/tag/6.2

closed 2016-12-22 by MoDuLah