TBXin/NFirmwareEditor #115: Nfirmware Editor button grayed out in Toolbox

Starbuck3733T, 2016-12-15:
I can't get the myEvic configuration button to do anything, my mod is plugged in and screenshooter works fine, but the button for myEvic config is not clickable.  the config editor in V5.1 still works. Running latest build of myevic (downloaded today 2016-12-15.)

Any ideas?

TBXin, 2016-12-15:
That's fine.

Starbuck3733T, 2016-12-15:
This is the expected behavior? Thanks.

(I guess I should try articfox...)

TBXin, 2016-12-15:
Yes, myEvic configuration utility is not developed yet

Starbuck3733T, 2016-12-15:
Thank you very much for the super quick answer.  I'll stick with 5.1 for now.

closed 2016-12-15 by Starbuck3733T

TBXin, 2016-12-15:
It's your choice, but I must admit that configuration utility from 5.1 is not designed for myEvic firmware. You can ruine all your settings

Starbuck3733T, 2016-12-15:
yeah i've done that a few times. I just now got temperature control dialed in and working right so I'm nervous to switch firmwares.  I'm using the 'sweet' algo for TC and I have idea if articfox will behave similarly.

great work on the project though!