TBXin/NFirmwareEditor #116: Vout2Batt.patch doesn't work on rx200

morganmachine91, 2016-12-17:
Issue occurs when trying to apply Vout2Batt or Vout - Battery patch. 

When either is applied, an error message is produced saying "Patches have not been installed due to conflicts," and listing the conflicting patch as whichever once was not selected. For example, if Vout2Batt is selected and 'Apply Patches' is clicked, it will fail saying there is a conflict with Vout - Battery even though Vout - Battery is not applied, and vice versa.

TBXin, 2016-12-17:
I do not answer questions about the ruined firmwares.
Before apply patches you should look at "Compatible" column, and i sure, it says "No".

closed 2016-12-17 by TBXin

morganmachine91, 2016-12-20:
Hey! Thanks a ton for the answer, didn't know I was asking something that I
shouldn't have. I'm not completely sure I understand your reply though, I'm
just trying to figure this out. Are the vout2battery patches just
incompatible with the rx200? My understanding was that the included patches
were all supposed to be compatible, but that you couldn't use some if you
were using others at the same time. If I read correctly, the
vout2battery.patch was compatible if Vout-Battery.patch was not installed,
and vice versa. I'm just wondering if there is any way to install one of
those, as having a readout of the lowest battery level would be much more
useful than seeing the current voltage my mod is firing at.

Thanks for the reply and I definitely want to express my appreciation for
your work! I really appreciate developers who give up their free time to
run open source projects like yours.

TBXin, 2016-12-20:
This patches used unused glyphs space.
I can't remember what concrete glyphs space they are using, but first off all you should apply patches and only than change graphics.

Take a pure rx200 firmware from wismec and you will see that you can apply one of these patches without any problems.

[image]

[image]

Here glyphs, that are unused by firmware, but their space are used to store patch data

[image]

And here is your problem, you are changed this glyphs in your firmware and now NFE tells you that patches are incompatible with firmware.